bigtreetech/BIGTREETECH-TFT35-V1.2 #2: V1.2.10 - missing bmp & fonts. Update isnt working.

DynaMight1124, 2019-03-22:
Hi

I saw you have updated the firmware to V1.2.10, however previously you had included bmp for icons and fonts too. When I tried to flash V1.2.10, it did flash fine but the screen icons were all interlaced. 

I flashed back to the previously firmware and everything worked fine. Is it because it's missing the bmp and fonts or an issue with the newest firmware?

nogarth, 2019-03-23:
i have same issue

merlin2105, 2019-03-26:
i have the same issue , whre i can get the icons.. or the previously Firmware...

merlin2105, 2019-03-26:
where can i build the bin file.... Atom?

DynaMight1124, 2019-03-26:
You can get the bmp/icon files plus the bin file for 1.2.8 from: https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/tree/d5f230e83f4f43cebb2cf70c5753167044b85454

ArturNadolski, 2019-04-02:
I have same issue.
I also try with original icon set from v1.2.8 but without luck (icon files in BMP folder at SD).

Msq001, 2019-04-24:
hello everyone, we have updated the firmware now, it include the "bmp","font" folder  and the newest  .bin firmware [here](https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/tree/master/firmware/Copy%20to%20SD%20Card%20root%20directory%20to%20update),