bigtreetech/BIGTREETECH-TFT35-V1.2 #4: Firmware update

elandesign, 2019-03-29:
Updating the firmware on my board has left the device unusable. On boot the calibration process starts (although with a blue background) and it consistently says "Adjustment failed, please try again"

Board version 1.2, firmware version 1.2.10

[image]

BaggaShite, 2019-04-19:
v1.2.10 firmware doesn't work. Seems nobody cares.

Msq001, 2019-04-24:
@BaggaShite it have be fixed 4 days before,  so sorry for that, you can download the newest .bin file from [here](https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/tree/master/firmware/Copy%20to%20SD%20Card%20root%20directory%20to%20update) and update it again