bigtreetech/BIGTREETECH-TFT35-V1.2 #5: Custom Icons

DynaMight1124, 2019-03-31:
If anyone is interested in some different icons. I created some: https://github.com/DynaMight1124/Bigtreetech-TFT35-Icons

Currently only work with firmware 1.2.8... since 1.2.10 doesnt work for anyone!

ArturNadolski, 2019-04-02:
Good job!

Msq001, 2019-04-24:
hi~ we have updated the icons [here](https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/tree/master/firmware/Copy%20to%20SD%20Card%20root%20directory%20to%20update)

valerie16901, 2019-07-30:
That's very cool, can you tell me what setting did you use for saving the bmp file, I have created some icons and exported in 16bit as the manual says, but they won't load :( I am using tft24 by the way.

DynaMight1124, 2019-07-30:
I used either 24 or 32 bit, rather than the suggested 16bit, cant remember exactly now. I had to try a few settings.

valerie16901, 2019-07-30:
Thank you! 32bit seems to do the trick, they should update the manual.

jinantha, 2019-08-27:
below is the only way worked for me,
ImageMagick 

magick convert Reserve_6.bmp -alpha set -define bmp:format=bmp3 -define bmp3:alpha=true Reserve_6.bmp

jinantha, 2019-08-27:
> Thank you! 32bit seems to do the trick, they should update the manual.

below is the only way worked for me,
ImageMagick

magick convert Reserve_6.bmp -alpha set -define bmp:format=bmp3 -define bmp3:alpha=true Reserve_6.bmp