bigtreetech/BIGTREETECH-TFT35-V1.2 #8: Where is a configuration file?

0Jmaster0, 2019-04-11:
Hello. I've bought the screen and now I'm trying to find out how to connnect it to my printer. The main problem is that my printer has two extruders, but I cannot find way to configue tft35. There is no such file on the Github. Where can I get configuration file or how to configue this screen?

Thank you for answer

ArturNadolski, 2019-04-12:
As far I know, there is no configuration file at Big Tree Tech TFT35. Configuration file has MKS TFT.

ArturNadolski, 2019-04-27:
I would like to make suggestion and add configuration file which can be easy change by user. Something similiar to MKS config file. It can be some mapping of definitions from configuration.h file when we can change temperature, speed, dimensions etc plus default language.