bigtreetech/BIGTREETECH-TFT35-V1.2 #9: BigTreeTech TFT35 & Raspberry Pi (Octoprint)

BakaBazooka, 2019-04-16:
Cannot get the TFT35 touchscreen to work with octoprint using a MKS GEN L?
When removing the touchscreen, octoprint works normally.

How to fix this problem

Msq001, 2019-04-19:
maybe you use a same UART port in TFT35 and Raspberry Pi,
The USB and AUX-1 use the same UART.  you can click  "Settings->Disconnect" on touch screen to disconnect the TFT35, or use the other UART connect with TFT35

BakaBazooka, 2019-04-19:
I am using a bundle of 2130 on SPI connections. 

Is there a way to separate the USB and AUX-1 connection??

Msq001, 2019-04-22:
[image]
yeah,  marlin-bugfix-2.0.x support the second serial port.  you can `#define SERIAL_PORT 0` for usb, and `#define SERIAL_PORT_2 2` for TFT35,   wiring RX(TFT35) to TXD2(MKS Gen L), TX(TFT35) to RXD2(MKS Gen L)

BakaBazooka, 2019-04-24:
I am already using the EXP1 & EXP2 pins for the RepRap LCD, and the AUX-2 & AUX-3 for the TMC2130 SPI connections. 
Is there a way to address or define the serial port to the AUX-1 on the MKS Gen L for the touchscreen?

Msq001, 2019-04-25:
the serial port in AUX-1 was be wired with USB in PCB, so if you use Raspberry Pi by USB, Serial ports on AUX-1 are Actually unusable.  atmega2560 has four hardware serial ports, two of which are used as endstops, leaving only two serial ports in AUX-1 and EXP-1 for use.
I want to know why TFT35 and RepRap LCD need to be used at the same time. T_T

BakaBazooka, 2019-04-29:
I am using the RepRap LCD to adjust my BLTouch settings, since the TFT35 doesnt come with BLTouch commands. So i was hoping to get it working with the two screens.

Has anyone done a step by step guide for Marlin 2.0 for Ender 3? 

Would this problem be solved by upgrading to the SKR V1.3?

Msq001, 2019-04-29:
The pin of RepRap LCD display is on EXP-1, and the EXP-2 is for SD Card,  So if you want to use EXP-2 to display RepRap LCD and EXP-1 to insert TFT35, you need to modify the `pins_xx.h` for MKS_GEN_L in Marlin firmware.

BakaBazooka, 2019-05-06:
Thanks for the information. I think i might order a SKR v1.3 which would have more serial ports for both screens.

Msq001, 2019-05-06:
The USB port of SKR V1.3 uses virtual serial port. not the same UART as TFT

MDRudnicki, 2020-04-17:
This worked for me in Configuration.h, Marlin 2.0.:

#define SERIAL_PORT 0
/**
 * Select a secondary serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 -1
I'm running a BTT SKR v1.3 + TFT3.5 + Octoprint and can control the printer through the touchscreen and Octoprint.

arut16, 2020-04-17:
@MDRudnicki You just connected your RPi by USB to SKR ?

bdelia, 2020-04-29:
@MDRudnicki , I have a similar setup except I am using the TFT24 V1.1 instead of the TFT3.5.  I have the serial ports defined as you do in Marlin, but I cannot seem to get Octoprint and the TFT to operate at the same time.  I am using the SKR V1.3 board...Just got the TFT24 a couple of days ago and stuggling to get it fully configured.

Would you mind sharing your Configuration.h, Configuration_adv.h, and aany files associated with PIN assignments like pins.h, pins_BTT_SKR_common.h and pins_BTT_SKR_V1_3.h?  I'd like to compare and see if I have missed something with the Marlin config.

MDRudnicki, 2020-04-29:
Yes

MDRudnicki, 2020-04-29:
Hi Brett,

Here are the files. I have not touched pins.

I found that the screen sometimes does not turn on when the printer is
switched on, but will come on after turning off and on again

I hope this helps.

Note- the configuration files are configures for a BLTouch, and also have
my extruder and hot end PID settings.

/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration_adv.h
 *
 * Advanced settings.
 * Only change these if you know exactly what you're doing.
 * Some of these settings can damage your printer if improperly set!
 *
 * Basic settings can be found in Configuration.h
 *
 */
#define CONFIGURATION_ADV_H_VERSION 020000

// @section temperature

//===========================================================================
//=============================Thermal Settings  ============================
//===========================================================================

//
// Custom Thermistor 1000 parameters
//
#if TEMP_SENSOR_0 == 1000
  #define HOTEND0_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND0_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND0_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_1 == 1000
  #define HOTEND1_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND1_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND1_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_2 == 1000
  #define HOTEND2_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
[... 1439 lines cut ...]
 * to allow leveling to avoid plate clamps on only specific
 * sides of the bed. With NOZZLE_AS_PROBE negative values are
 * allowed, to permit probing outside the bed.
 *
 * If you are replacing the prior *_PROBE_BED_POSITION options,
 * LEFT and FRONT values in most cases will map directly over
 * RIGHT and REAR would be the inverse such as
 * (X/Y_BED_SIZE - RIGHT/BACK_PROBE_BED_POSITION)
 *
 * This will allow all positions to match at compilation, however
 * should the probe position be modified with M851XY then the
 * probe points will follow. This prevents any change from causing
 * the probe to be unable to reach any points.
 */
#if PROBE_SELECTED && !IS_KINEMATIC
  //#define MIN_PROBE_EDGE_LEFT MIN_PROBE_EDGE
  //#define MIN_PROBE_EDGE_RIGHT MIN_PROBE_EDGE
  //#define MIN_PROBE_EDGE_FRONT MIN_PROBE_EDGE
  //#define MIN_PROBE_EDGE_BACK MIN_PROBE_EDGE
#endif

#if EITHER(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL)
  // Override the mesh area if the automatic (max) area is too large
  //#define MESH_MIN_X MESH_INSET
  //#define

bdelia, 2020-05-01:
Hey @MDRudnicki , thanks for this.  I may be missing it in the cut and paste, but did you also share your Configuration.h file?  I seem to only see the Configuration_adv.h file.  If you could also share the Configuration.h file that would be great.

Also, if you could attach them rather than cut and paste that might be easier.

Many thanks!

MDRudnicki, 2020-05-01:
Here are my configuration files

/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2019 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020000

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some standard links for getting your machine calibrated:
 *
 * http://reprap.org/wiki/Calibration
 * http://youtu.be/wAL9d7FgInk
 * http://calculator.josefprusa.cz
 * http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 * http://www.thingiverse.com/thing:5573
 * https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 * http://www.thingiverse.com/thing:298812
 */

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
// config/examples/delta directory and customize for your machine.
//

//===========================================================================
//============================= SCARA Printer ===============================
//===========================================================================
// For a SCARA printer start with the configuration files in
// config/examples/SCARA and customize for your machine.
[... 1591 lines cut ...]
 */
//#define SD_CHECK_AND_RETRY

/**
 * LCD Menu Items
 *
 * Disable all menus and only display the Status Screen, or
 * just remove some extraneous menu items to recover space.
 */
//#define NO_LCD_MENUS
//#define SLIM_LCD_MENUS

//
// ENCODER SETTINGS
//
// This option overrides the default number of encoder pulses needed to
// produce one step. Should be increased for high-resolution encoders.
//
//#define ENCODER_PULSES_PER_STEP 4

//
// Use this option to override the number of step signals required to
// move between next/prev menu items.
//
//#

Fabriglas, 2022-01-21:
Just setting up my skr 1.4 turbo with btt 3.5 e3 TFT ..and see the Pi may not be able to hook up for octoprint... any advice?