bigtreetech/BIGTREETECH-TFT35-V1.2 #10: Bootloader code

xvortex, 2019-04-19:
Any chance to see the bootloader code too?

omarcin96, 2019-04-24:
Could you add bootloader binary files to firmware folder?

xvortex, 2019-05-04:
Here we go - BIQU TFT35 bootloader binary: [link]https://mega.nz/#!E1N2TQZR!U0_q_gaPAaunFWr70RzyIVY5eieDWNxgQPJeHoc--Nw
Load address: 0x80000000

omarcin96, 2019-05-04:
thank you, it works.

Alex-TV, 2019-09-20:
I bought BIGTREETECH TFT35 V2.0. Connected the st-link to SWD. I flashed the firmware BIQU_TFT35_APP1_V2.0.23.bin via the st-link utility. Got a white screen. Then I flashed bootloader. Now I have a error, Illegal FLASH APP! Any help needed.

bullgun, 2019-09-25:
> 
> 
> Here we go - BIQU TFT35 bootloader binary
> among with IDA listing and disassembly for studying purposes: [link](https://mega.nz/#!MpcQFS4C!Ncn4hq9NG2_oB7ttJpxRRzKps5PaQzUY83hxzlBMKm4)
> Load address: 0x08000000
> 
> P.S. still waiting for source code...

How to burn the bootloader? Can i use a arduino? Thanks

bullgun, 2019-09-25:
I tryed to upload the firmware to the screen, it starts updating the fonts and the icons, , when it finish have the same firmware and doesnt update, its the bootloader?

ledfreak3d, 2019-09-26:
> I bought BIGTREETECH TFT35 V2.0. Connected the st-link to SWD. I flashed the firmware BIQU_TFT35_APP1_V2.0.23.bin via the st-link utility. Got a white screen. Then I flashed bootloader. Now I have a error, Illegal FLASH APP! Any help needed.

same thing here

Alex-TV, 2019-09-26:
I solved the problem. Now it works as it should.

ledfreak3d, 2019-09-26:
Yep ssme here working again flashed it back via platformio

my355vega, 2019-12-29:
I am having an issue, I did the flash threw sd card and it worked except now its showing icons that are half blank and the words are like in a dote matrix. how do I flash it back or does anyone have a boot-loader that works? I am trying to get to work on my skr mini e3 1.2 and not having any luck, thought maybe it was the screen but I guess its in the skr firmware.

Cooper19876, 2020-01-28:
How do you fix this issue its doing my head in,       "Illegal FLASH APP!"

my355vega, 2020-01-28:
What issues are you having?

Cooper19876, 2020-01-28:
I put this 👇to my tft35 V2.0 BIQU_TFT35_APP1_V2.0.23.bin  now my screen just says. Illegal FLASH APP!

Cooper19876, 2020-01-28:
Fixed it now I put buildroot on sd with  BIQU_tft35_APP1_V2.0.25.1.bin and tft35 and now it works

radek8, 2021-04-11:
@xvortex 
Hello
Can you please reissue the Botloader?
The previous link no longer works.