bigtreetech/BIGTREETECH-TFT35-V1.2 #11: V1.2.10 Doesn't Work"

BaggaShite, 2019-04-19:
Is this firmware ever going to be fixed or should we just throw these screens in the bin? (or return them to the seller for a refund?)

Seems they're more interested in releasing new boards than sorting the problems with existing ones.

DynaMight1124, 2019-04-20:
It's been updated now and works fine

r0xbeat, 2019-04-24:
"No Printer Attatched" message on the screen after updating to V1.2.10.  WTF is going on with this Firmware???

YanooStar, 2019-04-24:
Same thing!

themaskedbear, 2019-05-04:
I have the same problem.
Someone know if there is a solution ?

boucz, 2019-05-04:
same for me... "no printer attached" on screen

themaskedbear, 2019-05-04:
Well,

I've found why "no printer attached" where were displayed.
For my part I was upgrading using the already compiled firmware for the skr v1.3 firmware provides here.

After compiling my own firmware with my configuration (ex: TMC2208 UART used) the message disappeared.

Note that, after upgrading on your own firmware, at every boot of your printer, this message will still be displayed until the boot is properly finished.

r0xbeat, 2019-05-05:
> 
> 
> After compiling my own firmware with my configuration (ex: TMC2208 UART used) the message disappeared.
> 

Which file can be configurated?

boucz, 2019-05-05:
I also recompiled the kernel (no change to any file from the git repo, as my board is a v1.2)
It now runs flawlessly :)
my bin file is 1Ko bigger (maybe lib version differs or anything else)

[BIQU_TFT35_V1.2.10.bin.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/files/3145426/BIQU_TFT35_V1.2.10.bin.zip)

boucz, 2019-05-05:
for other versions :

[BIQU_TFT35_V1.0.10.bin.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/files/3145432/BIQU_TFT35_V1.0.10.bin.zip)
[BIQU_TFT35_V1.1.10.bin.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/files/3145433/BIQU_TFT35_V1.1.10.bin.zip)