bigtreetech/BIGTREETECH-TFT35-V1.2 #24: After updating firmware to 1.2.11 my laptop doesn't recognise the skr v1.3 board

Loadie150, 2019-05-28:


Msq001, 2019-05-29:
the tft35 screen connect to skr v1.3 by uart interface, and the pc connect to skr v1.3 by USB,  these two way use different interfaces, so there should be no interaction between them.  have you update the marlin firmwre in skr V1.3 ?

Loadie150, 2019-05-29:
Yes, My laptop can send the info to the skr board but I can’t connect to the board with the USB and pronterface. Sent from  <https://go.microsoft.com/fwlink/?LinkId=550986> Mail for Windows 10 From:  <mailto:notifications@github.com> Msq001
 Sent:  May 28, 2019 6:15 PM
 To:  <mailto:BIGTREETECH-TFT35-V1.2@noreply.github.com> bigtreetech/BIGTREETECH-TFT35-V1.2
 Cc:  <mailto:danfraser@shaw.ca> Loadie150 ;  <mailto:author@noreply.github.com> Author
 Subject:  Re: [bigtreetech/BIGTREETECH-TFT35-V1.2] After updating firmware to 1.2.11 my laptop doesn't recognise the skr v1.3 board (#24)the tft35 screen connect to skr v1.3 by uart interface, and the pc connect to skr v1.3 by USB,  these two way use different interfaces, so there should be no interaction between them.  have you update the marlin firmwre in skr V1.3 ?

Msq001, 2019-05-29:
Can you see Marlin USB serial (COM xx) in the device manager of your computer?  if not,  Please down the newest marlin firmware from [Marlin official website](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x). and modify with reference to this [video](https://www.youtube.com/watch?v=oaXfXkPYHpw) You can fast-forward directly to the place where firmware is modified.