bigtreetech/BIGTREETECH-TFT35-V1.2 #26: Marlin menu

ArturNadolski, 2019-05-29:
Any chance for option to switch beteween TFT menu and Marlin menu?

dianlight, 2019-05-30:
It's a good idea but I don't know if it can be implemented due to the nature of the TFT35.
Since the TFT35 is a device that commands the motherboard via serial link, it does not receive from Marlin the information necessary to render the menu and execute the menu.

In any case, if nobody implements it before, I inserted it in the todo list of the things I will try to implement [in my branch fork.](https://github.com/dianlight/BIGTREETECH-TFT35-V1.2/wiki/Roadmap)

Stay connected...

Gerggg, 2019-06-12:
Hello, it's a great idea, the tft screens are nice but from a point of view function, they are very poor. Having the richness of Marlin screens on tft would be a revolution in terms of interface on 3D printers.
Good luck Dianlight !!

Gerggg, 2019-06-13:
Hello, do you know the Extensible UI API? it is obviously used for TFT screens, it should help you move faster.
On a post in the marlin project, there is one person who gave me an example of use for a Creality printer.

https://github.com/InsanityAutomation/Marlin/blob/CrealityDwin_2.0/Marlin/src/lcd/extensible_ui/lib/Creality_DWIN.cpp

ArturNadolski, 2019-06-17:
TFT24 is working this way (Marlin menu). I assume that software can be ported to TFT35.

Gerggg, 2019-06-18:
Really? can you give a link to the firmware of the TFT24? Thanks in advance.

ArturNadolski, 2019-06-18:
I just saw it here: https://www.facebook.com/100021425589762/videos/397007831023400/
For a now, there no official FW source on GitHub.

Gerggg, 2019-06-18:
Thank you, it seems that the Bigtreetech TFT24 does not exist yet, it should be released this week, I imagine that BTT will also release the firmware in the week.