bigtreetech/BIGTREETECH-TFT35-V1.2 #28: Issue importing icons

xoniq, 2019-06-01:
Hello,

I'm running the TFT35, but tried to create custom icons. But my icon become distorted when being imported into the screen.

I'm using a Mac, so I dont have much options for BMP stuff. So I create it as a PNG, and use ImageMagick to convert it to BMP without alpha. The colors are fine, it tries to import, but the image is imported as show below.

How can I solve this?

[image]

bleughb, 2019-06-01:
Upgrade to the latest firmware

xoniq, 2019-06-01:
Already using it. V1.2.11.
That's the version where I have this issue.

Msq001, 2019-06-03:
Is the pixels of your icon is 95*95?

dianlight, 2019-06-04:
Hi,
 di you found a solution? I'have the same problem on a Mac with ImageMagic.

xoniq, 2019-06-04:
@Msq001:
Yes, exactly that size. Weird part is; the boot screen is correct, icons all distorted.

@dianlight:
No haven't been able to fix...

Msq001, 2019-06-05:
@dianlight @xoniq  can you send me a icon that I can check it

Msq001, 2019-06-05:
my e-mail: 3001154986@qq.com

Msq001, 2019-06-06:
So sorry for that,  I've confirmed that this is a bug decoding BMP file in the bootloader,  I'm thinking about moving the code of decode BMP and bitmaps fonts from bootloader to app code. and fix this bug,   before this, you can avoid this bug by setting the bit depth of the icon to 32 bits instead of 24/16 bits. as shown in the following picture.
[image]

xoniq, 2019-06-06:
@Msq001,

Thanks, will try it at 32 bit tonight, if it doesn’t work I’ll send the BMP file in a zip file here in a reply (so it doesn’t get altered)

dianlight, 2019-06-06:
It works for me! But in reality it is not enough to switch to 32bit it is also necessary to check that ImageMagick in the passage does not pass from bmp v3 to bmp v4.

This is the command I used to "fix" the Reserved_6.bpm

Before:
```
$ file Reserve_6.bmp
Reserve_6.bmp: PC bitmap, Windows 3.x format, 95 x 95 x 24
```

Transformation:
` $ convert Reserve_6.bmp -alpha set -define bmp:format=bmp3 -define bmp3:alpha=true Reserve_6.bmp`

After:
```
$ file Reserve_6.bmp
Reserve_6.bmp: PC bitmap, Windows 3.x format, 95 x 95 x 32
```

xoniq, 2019-06-07:
@dianlight Thanks for the suggestion, i'm gonna try it in a few hours. Already tried one file.

**Before:**
```
Fast.bmp: PC bitmap, Windows 98/2000 and newer format, 95 x 95 x 32
```
**Command:**
```
convert Fast.bmp -alpha set -define bmp:format=bmp3 -define bmp3:alpha=true Fast.bmp
```
**After:**
```
Fast.bmp: PC bitmap, Windows 3.x format, 95 x 95 x 32
```
I will post my results here in a few hours.

xoniq, 2019-06-07:
@dianlight @Msq001

The solution of @dianlight was the key (THANKS!). Now all images are perfectly imported.

This issue can be closed with this information for macOS and Linux users.

closed 2019-06-30 by xoniq