bigtreetech/BIGTREETECH-TFT35-V1.2 #30: Flip screen upside down

macasero, 2019-06-06:
Is it possible to flip the screen upside down, so we have the SD on left position other than right position?

mpallen509, 2019-06-07:
Also interested in flipping the GUI. Have not been able to identify any built in rotation commands.

SlyKittyCat, 2019-06-07:
been waiting for this my screen is collecting dust on my desk also baby stepping z offset would be nice

mpallen509, 2019-06-07:
Should be capable of baby stepping as there is a menu item for it during printing. Settings are found in config_adv. Make sure to enable them on the printers board first, then the screen should stop displaying unknown command errors.

ArturNadolski, 2019-06-20:
void set_display_direction(u8 direction)
{

	if(direction) {
    LCD_WR_REG(0x36); 
    LCD_WR_DATA(0xE8);
    Delay_ms(120); 
	} else {
    LCD_WR_REG(0x36); 
    LCD_WR_DATA(0x48);
    Delay_ms(120);     
	}

}
0-landscape, 1-portrait etc
Take a look at driver manual: https://www.elecrow.com/download/ILI9488%20Data%20Sheet_100.pdf
chapter 5.2.30. Memory Access Control (36h) and MX MY MV

mpallen509, 2019-06-20:
@ArturNadolski Thanks for the function, although being a noob at C&C++ I am still lost on how to implement this. 

If you would be so kind, where would I insert the function (GUI.c, ui_draw.c, etc...) and should I be assigning a hex value to 'direction?' (0x01, 0x02, etc...)

- Any help is appreciated thanks for the tip already.

ArturNadolski, 2019-06-20:
Flipping the screen is quite simple but flipping touch screen needs more coding.