bigtreetech/BIGTREETECH-TFT35-V1.2 #32: After firmware update icons and text are not correctly displayed.

MercadoNegro, 2019-06-10:
Hello,

I have a 3.5 TFT v1.0, it wasn´t connecting to my board (MKS Sbase v1.3), then I updated the firmware. After the firmware update most icons are not correctly displayed, and boot image is wrong.
I downloaded the new firmware source and compiled it without issues, but after flashing it the result is the same.

https://photos.app.goo.gl/YTgXiDu3L1Xk56zi9

https://photos.app.goo.gl/coF2LkGAPb696Qa39

https://photos.app.goo.gl/3LNMQPpZ5aFNzF2c6

I already checked the icon format, they are BMP v3, 95x95 and 32 bit. (the default on the GitHub repository).

I don´t know what else to do.

Thanks.
Gustavo

Msq001, 2019-06-19:
oh, This is because the bootloader is different between v1.0 and  v1.1/v1.2. Now I'm transferring the code of updating icons and bitmap fonts from bootloader to app, which will solve this problem when I finish.

MercadoNegro, 2019-06-19:
Thanks. Will wait for the solution.

MercadoNegro, 2019-07-07:
Do you know when you will have this issue fixed? It´s still not working for me.

NightPostal, 2019-07-09:
Im having the same issues with the Graphic now showing correctly. Have DOWNLOADED the v 1.2 and not working.

Msq001, 2019-07-10:
@MercadoNegro  I'm working on integrating 3.5, 2.8 and 2.4 inches in one source code. and just need modifying config to generate correct firmware,  So it took a long time， It is scheduled to be completed before 2019.7.16 (china time), and I will upload the firmware then.  It will be a workable firmware.  I'm sorry to have kept you waiting so long.

MercadoNegro, 2019-07-10:
Oh! That would be great!! I thought it was a minor change, but I see you are doing a totally new approach.
Thanks for your help and patience. Will wait.

Msq001, 2019-07-16:
hello， the source code have be uploaded in here https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware, Because it contains different hardware versions such as TFT35 V1.0 V1.1 V1.2 TFT24 V1.1  TFT28, So I think it's inappropriate to keep it in this repository, so I create a new repository for it.
If you have a TFT35 V1.0， you need only copy folder “TFT35” and “BIQU_TFT35_V1.0.20.bin” to SD card root dir， and insert to TFT，then reset TFT， it will updating by itself， Please delete these file in SD card after you have updated successful，After that, TFT will work properly.