bigtreetech/BIGTREETECH-TFT35-V1.2 #40: Flashing Board

MrShoe1980, 2019-07-04:
After I flashed the board with the current firmware, It asks me to calibrate the red dots.  I click on the left right and bottom right.  Then a center black dot appears, if I touch it it fails, if I do nothing it fails...

Msq001, 2019-07-10:
Please try to click on the center of the new point. The specific steps are to 
`click on the red dot in the upper left corner` - > 
`click on the red dot in the upper right corner` - > 
`click on the red dot in the lower right corner` - > 
`click on the black dot in the center of the screen` - >  calibration success.

Msq001, 2019-07-10:
If you still fail in calibration,  Note that the coordinates at the top left of the screen when the get fails, (240,160) are the perfect values,  and if it's (0,0), it's a hardware issue of screen.
Please click as far as possible on the center of the point.  If possible, sharp objects can be used for calibration to get more accurate values