bigtreetech/BIGTREETECH-TFT35-V1.2 #41: Beautiful Even Fonts for Western and Cyrillic

Acenotass, 2019-07-04:
I want to share the fonts that I created in return for the original.
[image]
Original
[image]
Original Fixed
[image]
Coder Font
[image]
Ubunru Font
[image]
Helvetica Thin
[image]
Helvetica Thin Inalic

ArturNadolski, 2019-07-04:
Good job!
Could You please describe font creation process? Any tool for generation font file?

Acenotass, 2019-07-04:
> Good job!
> Could You please describe font creation process? Any tool for generation font file?

I use SG Bitmap Font editor
http://corosan.ru/sgfed/

[FONT-Templates.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/files/3358966/FONT-Templates.zip)
To much time to explain, just use the attache templates.

xoniq, 2019-07-04:
Very nice! Thanks! I’m using the Ubuntu font.

thisiskeithb, 2019-11-02:
@Acenotass: Any chance you can help/advise on what to change to get these fonts working with [recent changes](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/89) from BigTreeTech?