bigtreetech/BIGTREETECH-TFT35-V1.2 #42: Print randomly stops Tft35 v1. 1

Waynan, 2019-07-07:
I'm having some strange behaviour where if printing from the sdcard the print will randomly stop trying to enter any menu on the tft results in it becoming completely unresponsive. Hitting reset results in no printer connected condition, only a full power cycle allows me to resume the print.

I'm running marlin 2.0 bug fix on an skr v1. 1, I'm also running the latest release of the tft3.5 v1.1 firmware. 

Has anyone else had this issue?

Adik22, 2019-07-11:
Have the same problem. Using tft35 with SKR v1.3. Yesterday printing with no problems, about 10h print time. But who know what tommorow brings :/

xoniq, 2019-08-01:
I thing the connections from the TFT35 isn't stable enough for printing. Had the same issues, also then I pickup the TFT it restarts the printer, so there are some issues in the wiring for me.

I changed Marlin config to print from the SD inside the SKR 1.3. Less external connections, less chance to fail.

Adik22, 2019-08-01:
I did the same, now always printing with onboard sd card and using stock Ender 3 screen.

jinantha, 2019-09-05:
Hi Same here,
Didn't see this thread and posted a new issue  [https://github.com/bigtreetech/BIGTREETECH-TFT35-V1.2/issues/48](url) of my own.
No response from BigTreeTech.. too bad :(
I went back to the stock CR10 display.
It had no issues and have more options.

Hoangdai4017, 2019-09-13:
Same here.
:((

https://youtu.be/HnCBxA3adFA?list=LLvvX8gJoIP1svSnOkJSbHcg&t=156

jenschalupka, 2019-10-03:
Hi, just to let you know they moved their repository to https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware . There they documented the fix for that with Vx.x.21. Current Version is Vx.x.23.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/README.md

cadu398, 2019-10-08:
I am having the same problem because I can not update tft35 already tried to change the sd card and does not exit 2.0.20 already tried to change the file name to what is on the site and nothing

cadu398, 2019-10-08:
I've been researching and saw that the tft35 has a problem that can not be upgraded has to have a st-link and make more fun to work anyone knows where I can download a version of marlin2.0 bugfix that can print by sd onboard this new is not supporting reading error

cadu398, 2019-10-08:
I've been researching and saw that the tft35 has a problem that cannot be upgraded has to have a st-link and make a bigger joke to work does anyone know where I can download a version of marlin2.0 bugfix that can print by sd onboard, this new not is supporting the reading error

lbarrettanderson, 2019-11-02:
@iChalix I'm on x.x.24 and still have the issue. Has anyone been able to update and have the issue fixed?