bigtreetech/BIGTREETECH-TFT35-V1.2 #44: Delay TFT35 boot

mpallen509, 2019-07-16:
Wondering if there is an effective way to delay the booting of the TFT35 v1.2 GUI. (maybe even a simple wait command during the boot screen.)

This would be so the control board (SKR v1.3) has enough time to boot prior to the screen. 

The main reason for this is to help remove the time-frame in between the screen being operable and it still waiting for the board communication (i.e. No Printer Connected)

SISLANGER, 2019-07-16:
Depending on version using
either main.c
startUpTime < 300
300 is in ms so change that

or 
in mode.c
while(OS_GetTime() - startUpTime < 300)

try updating both board and TFT as I have no problem with "No Printer Connected issue" with or without timer change. I did have on a few versions back of the TFT and Marlin Firmware

mpallen509, 2019-07-16:
Just updated both, still getting some "No Printer Connected" during the boot process. It fades after a few seconds, but for user experience I wanted the control board to be ready by the time a user starts tapping the screen. 

May try increasing the startUpTime or effectively the boot screen length. (I don't know how I missed that code when I was looking for some delays!)

SISLANGER, 2019-07-16:
LOL code blindness it happens, 

It causes a pause showing the logo for longer so you can count the longest it takes the main board to load about 4000 - 7000 ms should do it

mpallen509, 2019-07-16:
Ya I will attempt it and see what my team thinks of the delay. 

Already changed the "No Printer Connected" to "Please wait..." so it is a little less alarming for 3D printer newbies. 

Thanks for the tips @SISLANGER

closed 2019-07-16 by mpallen509

SISLANGER, 2019-07-16:
Please wait is a better way to convey I may even change that myself. The code works I have it on mine to show the custom longer for a bit longer.

mpallen509, 2019-07-16:
Ya it is a super easy change that communicates a little more effectively. 

Depending on language preference, you can find it in the 'language_en.h' under the EN_UNCONNECTED definition.

SISLANGER, 2019-07-16:
Thank you will have a dig in there later