bigtreetech/BIGTREETECH-TFT35-V1.2 #46: Can't read onboard SD card on SKR 1.3

Dendrowen, 2019-07-29:
I am unable to read the SD card in the SKR 1.3 from the TFT. It says:

Loading....
Read SD card error!

How can I solve this in the marlin firmware?

xoniq, 2019-08-01:
I guess it's because the SD card isn't initialized on start. I'm using both the TFT35 and the Smart controller LCD, but I just print from the Smart controller LCD because I have the init SD from that LCD, to mount the SD which is placed in the SKR 1.3.

melufre, 2019-08-01:
hello
 @xoniq  could you post what you change on marlin cfg in order to be able to print from On board SD card

i have the same feeling as you about the stability of printing from TFT35 SD card

xoniq, 2019-08-01:
Hi @melufre

Here are my current config files:
https://github.com/xoniq/Marlin/tree/SKR1.3_TMC2130/Marlin

Dendrowen, 2019-08-01:
The card is initialized because I can see it from my PC. Not from the TFT though. It doesn't matter if the PC is connected or not.

xoniq, 2019-08-01:
If you can see it on the PC, it means the PC mounts the card, not the board itself. With no PC attached and only a screen, I have to manually mount it.

Dendrowen, 2019-08-01:
And how can I achieve that?

xoniq, 2019-08-01:
Gcode:
http://marlinfw.org/docs/gcode/M021.html

Make sure you have enabled that Marlin should use on-board SD by default.

REDxFROG, 2019-10-10:
I think the BAUDRATE has to be at 115200

so

#define BAUDRATE 115200

then it should work. But better confirm the firmware has been written/changed.