bigtreetech/BIGTREETECH-TFT35-V1.2 #48: Marlin V2.0 on SKR V1.3 with TFT35 v1.2 printing halts.

jinantha, 2019-09-04:
My hardware;
Ender 3, 
BigTreeTech SKR V1.3, 
TMS2130 - BigTreeTech v3, 
BigTreeTech SKR TFT35 V1.2 
latest Marlin 2.0 bug fix (downloaded yesterday 2019-Sep-03). 
Printer just halts around 1st or 2nd layer.
I had to power cycle.
I was using the LCD SD card option to print.
`#define SDCARD_CONNECTION LCD`

Then I changed to `#define SDCARD_CONNECTION ONBOARD`
And it worked!!!

So conclusion to me would be printing over LCD module might be the cause for halting.
Is this a bug?
Now I have a unusable LCD module and cannot return for exchange or refund. Bigtreetech store in Aliexpress not responding.
Anyway I'm using the stock CR10 display and printing from onboard SD.
No issues so far.

Config files and pictures -> [https://www.dropbox.com/sh/vhlia1qlqtfvovi/AACfHDK3q7LXMsKnw1vZlpQ9a?dl=0](url)

pisczek, 2019-09-18:
I am having the exact same issue. I have been using the stock LCD and the onboard SDCARD and I have had no halting issues. I used to have them on every print while using the TFT.

primeeveil, 2020-02-26:
I'm having the same issue with larger / longer prints. I have noticed that I can still send commands to my skr 1.3, after the tft35 v2 halts, with my original ender lcd screen if I have also have it plugged in, which indicates it the TFT35 that is getting stuck.

ArxGene, 2020-05-28:
Hello, do we have any update on this issue yet? I am using the same setup and I have this problem. Some prints will finish up fine but other just not. The printer is randomly freezing and the temperatures remain stuck turned on. Wasted big amounts of materials and am super annoyed