bigtreetech/BIGTREETECH-TFT35-V1.2 #49: connecting to MKS SBASE

mpieroway, 2019-09-22:
the connector on the SBASE 1.3 is the same serial connector as the discount controllers, but the 5 leads form the Bigtreetech TFT 35 V2.0 don't fill the serial port. How do I connect the TFT to the SBASE?

mpieroway, 2019-09-23:
Found my answer on reddit. 

Essentially connect the 5V, GND, TX and RX to the four pins in the serial connector labeled 5V, GND, PO.2 and PO.3. The odd lead left is the reset which was then attached to a 6 pin connector close to the USB and Ethernet ports. 

Wish me luck, the next phase is the update of Marlin.

hamili, 2019-10-30:
any good news
is bigtreetech tft35 compatible with mks sbase board?

ledfreak3d, 2019-10-30:
Nope

mpieroway, 2019-10-30:
I do have it up and running. Its plugged into the serial connector where the mks touch screen attaches....I will upload an image later of pin config.

yevhenii-shch, 2020-09-10:
> I do have it up and running. Its plugged into the serial connector where the mks touch screen attaches....I will upload an image later of pin config.

can you publish the config?